HengyiWang/Co-SLAM #1: Releasing Code

Lilyo, 2023-04-30:
Hi @HengyiWang, 

Thank for your excellent work!, It's so impressive.

Any plan to release code for us to evaluate you model or provide a simple code for video real-time inference ?

Look forward to your reply!

HengyiWang, 2023-05-04:
Hi @Lilyo ,

Thanks for your interest in our work. I am currently in the process of cleaning up the code base and I will try to release the code by the end of next week:)

Lilyo, 2023-05-12:
Hi @HengyiWang , 

Thanks again for the great work, I'm also visualizing the output mesh live and it looks great!

[image]

glaciomarine, 2023-11-23:
> 你好@HengyiWang,
> 
> 再次感谢您的出色工作，我还以实时方式可视化输出网格，它看起来很棒！
> 
> [image]

你好，如何得到可视化网格，可以提供一下代码参考吗