HengyiWang/Co-SLAM #2: Queries in the SDF Loss Computation

manthan99, 2023-05-16:
Hi

First of all thank you for the wonderful work and for releasing the code-base. I had some queries regarding the implementation of the SDF loss which seems slightly different than that mentioned in paper. It would be great to know the reasoning behind this-

1. https://github.com/HengyiWang/Co-SLAM/blob/dfb69c80beff76492309ffa626c5865ba8d8f0be/model/utils.py#L142
In this case while computing the free space loss, it seems like the loss is being computed with respect to 1 and not w.r.t the truncation value. Is there a multiplication with the truncation value missing in the equation ?

2. While computing the masks for sdf and freespace, is there any particular reason why only the points in the front are considered for free-space loss and not the points in the back ? This seems a bit different then the paper where everything which is not in the truncation region is considered as free space.

Thank you

Best
Manthan

HengyiWang, 2023-05-16:
Hi @manthan99,

Thank you for your interest in our work. I'll address each of your points below:

1&2. The reason behind this difference is that we aim for the output of the MLP to be in the range of [-1, 1] instead of [-truncation, truncation]. In other words, the network predicts a scale factor or percentage rather than the truncation value itself. Therefore, for the free space loss, we use a value of 1. And for the truncation region, we need to multiply the truncation.

3. This is because Co-SLAM focuses on indoor scene reconstruction. In such scenarios, a single ray might pass through multiple rooms or objects. If we consider points in the back as free space, it might conflict with the actual scene geometry. Therefore, we only consider the points in the front as free space.

I hope these clarifications address your concerns. If you have any further questions or need more information, please feel free to ask :)

manthan99, 2023-05-16:
Hi @HengyiWang 

Thank you for the quick response. This makes it quite clear.

I had an additional question regarding the computation of the weights  (fs_weight and sdf_weight). Could you please provide some intuition behind this. 

https://github.com/HengyiWang/Co-SLAM/blob/dfb69c80beff76492309ffa626c5865ba8d8f0be/model/utils.py#L106

Let's say for example in the case where all points are lying in the truncation region, then the sdf_weight will be zero in this case. So we won't get any SDF loss in this case ?

Thank you 
Best
Manthan

HengyiWang, 2023-05-16:
Thanks for your question @manthan99. The computation of the weights (fs_weight and sdf_weight) aims to address the issue of imbalance between free space and SDF samples during training. 

Let's consider an example where we shoot a ray of 5m in a room-scale scene, with a truncation value set to 10cm. In such a scenario, the majority of the sampled points would likely fall within the free space region. This significant imbalance between free space and SDF samples can cause the network to bias towards predicting all points as free space, neglecting the SDF information.

Regarding your concern about the SDF weight being 0, it is right. However, this will not happen in our setting as our near-far range (usually 3-8m) is significantly larger than truncation.

manthan99, 2023-05-17:
Thank you for clearing my doubts.

closed 2023-05-17 by manthan99

iAhmedMaher, 2023-07-06:
Hi @HengyiWang,

Following up on your earlier comment on the truncation value

> 1&2. The reason behind this difference is that we aim for the output of the MLP to be in the range of [-1, 1] instead of [-truncation, truncation]. In other words, the network predicts a scale [...]

My understanding is that the SDF value outputted from the network is equivalent to (actual SDF value * truncation). If this is the case, shouldn't the equation for obtaining the weights be sigmoid(SDF) instead of sigmoid(SDF/truncation) in this line of code to follow the equations of the paper?

https://github.com/HengyiWang/Co-SLAM/blob/49cf9e6f947b7e13d453eb8070dee3974d959a50/model/scene_rep.py#L74

I tried removing the division by the truncation value, but the output mesh is completely wrong, so I would hope you help my understand why.

HengyiWang, 2023-07-06:
Hi @iAhmedMaher, that is a really good question, in the rendering pipeline of neuralRGBD, the SDF is always in the normalized space. You can probably check the response of the author of neuralRGBD [here](https://github.com/dazinovic/neural-rgbd-surface-reconstruction/issues/12).