HengyiWang/Co-SLAM #6: When to release data capture by ipad / iphone?

YiChenCityU, 2023-05-26:
Thanks very much.

HengyiWang, 2023-05-26:
Thank you for your interest, I will try to release the tutorials and the related code for capturing data using ipad/iphone by the end of this week.

HengyiWang, 2023-06-11:
Hi @wing-kit, it is great to see that you manage to capture data using iPhone/iPad and get the reconstruction results using Co-SLAM. 

Regarding your question about obtaining more photorealistic results, I have some suggestions for you to try:

1. In the latest commit of Co-SLAM, I added an option for rendering surface color. You can give it a try and see if it improves the photorealism of the reconstructed mesh. However, please keep in mind that the viewing direction is currently set to be the surface normal. If the actual viewing direction significantly deviates from the surface normal, the color might appear worse.

2. Co-SLAM performs joint optimization of both geometry and color. To enhance the color quality, you can experiment with increasing the weight of the RGB loss or increasing the number of mapping iterations (`iters: 10`). Note that the current hyper-parameter setting focuses on computational efficiency and uses only 5% of pixels (`n_pixels: 0.05`) from each keyframe. You can consider increasing this percentage to obtain more pixel observations

3. By default, the voxel size for mesh extraction in Co-SLAM is around 3cm, which might be slightly large for small objects. Reducing the voxel size further may help improve the small objects in the reconstructed mesh.

I hope these suggestions help you in achieving more photorealistic results with your reconstructed meshes. Let me know if you have any further questions or need assistance with anything else:)

mohcenaouadj, 2024-05-23:
@wing-kit Hello,

I'm trying to use Co-SLAM with PolyCam dataset yet I crumble a little bit with the configuration of the dataset and eventually the method, can you share with me the configuration you used, and any tips on how to prepare your dataset ?

Thank you