HengyiWang/Co-SLAM #11: Runtime on Replica

huanggan52, 2023-06-29:
Hello, In the paper, you mentioned that "For experiments with default settings (Ours), which runs at 17 FPS on the Replica dataset." However, I am experiencing a significantly lower runtime of only 3-4 FPS with approximate similar configurations. Is there anything else that needs attention? Looking forward to your reply and guidance.

JingwenWang95, 2023-06-29:
Hi @huanggan52 Thanks for your interest in our work! First of all, the 17 FPS we reported is the theoretical largest throughput our method can achieve, i.e. exluding the time required for data loading, delays caused by multi-processing with single GPU, etc. For actual run-time on Replica sequences with a single GPU, we've created a detailed benchmark here: https://github.com/JingwenWang95/neural_slam_eval/tree/main/benchmark/replica Co-SLAM can process a Replica sequence with ~3min20s.

Also may I ask how did you obtain the number of 3-4 Hz? Maybe you can provide some details about your machine and the settings you run Co-SLAM? Ideally, we need 2 GPUs to maximize the run-time performance, 1 for tracking and 1 for mapping. 

Best
Jingwen

hjr37, 2023-07-02:
Thanks for your surprising work, I met the same problem, while I run the original code without any modification. I tested the released code, specifically the single-thread version 'coslam.py', on the Replica room-1 dataset with ~12min43s. My supported device is RTX3090. I'm confused about what causes the efficiency difference between us. I would be grateful if you could provide an analysis.

JingwenWang95, 2023-07-03:
Hi @hjr37 Thanks for your feedback! It's quite weird... On our side we tested with both RTX3090ti and RTX2080ti and the run-time required to process a Replica sequence is around 3min20s and 4min20s respectively both including the time for saving meshes. So there must be something happening in your experiments. There are actually several factors (CPU, RAM, SSD, etc.) that might affect the run-time performance which is mainly down to the dataloading processing. For example if you store the data on slow HDD instead of fast SSD then the dataloading will be a bottleneck. Here we can provide some datails on the machines we ran our tests on: 

The RTX3090ti machine: intel Core i7 12700k CPU + RTX3090ti 24GB + 128GB RAM + Samsung 870QVO SSD
The RTX2080ti machine: intel Core i7 9700k CPU + RTX2080ti 11GB + 128 GB RAM + Samsung 870QVO SSD

Also have you swithed off all other optional utilities such as on-the-fly visualization? Is your CPU heavily used by other processes? Maybe can you provide some more details about your machine and how you ran the experiments? Some screenshots would be great.

closed 2023-07-21 by HengyiWang

zjulabwjt, 2023-10-12:
> Thanks for your surprising work, I met the same problem, while I run the original code without any modification. I tested the released code, specifically the single-thread version 'coslam.py', on [...]

Have you sovled this problem?And the runtime can arrive in result?