HengyiWang/Co-SLAM #12: freeze seem does not work?

ZZy129326999, 2023-07-03:
`
    def freeze_model(self):
        '''
        Freeze the model parameters (Mapping)
        '''
        for param in self.model.embed_fn.parameters():
            param.require_grad = False
        
        for param in self.model.decoder.parameters():
            param.require_grad = False
`

Here **require_grad** should be **requires_grad**, is it necessary to unfreeze after optimizing the pose? Or has it not been frozen, and has been jointly optimized?

HengyiWang, 2023-07-03:
Hi @ZZy129326999, thank you for reporting this. I forget to delete `freeze_model()` in the released version. Ideally, yes, you need an `unfreeze_model()` to unfreeze the model. However, in our case, the model params have not been jointly optimised as only the pose optimiser has been used during tracking. Please see below.

https://github.com/HengyiWang/Co-SLAM/blob/49cf9e6f947b7e13d453eb8070dee3974d959a50/coslam.py#L479

I have updated the code to remove this function that actually does nothing... Please feel free to contact us if you have any further questions:)

ZZy129326999, 2023-07-03:
> 

Thank you very much !! I learn a lot by your perfect work of which code format is very good. Look forward your futher work!

closed 2023-07-03 by ZZy129326999