HengyiWang/Co-SLAM #13: Runtime analysis Tables 1 & 2 in main paper

random-guest, 2023-07-04:
Dear Authors,

Thank you for sharing this outstanding work!

I was wondering how can we separate the overall runtime that you define below:

https://github.com/JingwenWang95/neural_slam_eval/tree/main/benchmark/replica 

as:

"The runtime is measured as the total processing time from the program starts until the final checkpoint is saved."

into Tracking, Mapping, and FPS as shown in Tables 1 and 2 in the main paper.

Thank you in advance for your time and effort in answering my issue.

Regards,
Karim,

JingwenWang95, 2023-07-04:
Hi @random-guest Thanks for your interest! Let me try to clarify the differences between those "run-time results".

So first of all, there are several implementation-specific details that might affect the fairness of the comparison, such as different dataloading, single vs. multi-processing. For example, iMAP measures their run-time of tracking and mapping with two separate GPUs while NICE-SLAM runs with a single GPU. For a fair comparison, we only measure the time needed to perform a complete tracking/mapping cycle, i.e. the time elapsed within the optimisation loop, using single-processing and on a single GPU. That is how we obtain the run-time in milli-second in Tab. 1 and 2 in our paper and also in our benchmark.

Regarding the FPS, in our paper we report the theoretical throughput under a perfect multi-processing scnerio, then we are able to compute the throughput FPS according to the run-time ms we obtained. Take Co-SLAM running on Replica for an example: one tracking cycle takes 58ms, one mapping cycle takes 98ms, mapping runs every 5 frame. This yields a throughput of 1000/58 = 17.4 Hz. However, for NICE-SLAM it will be 5 * 1000 / (78 * 5 + 5470), as the mapping appears as a huge bottleneck.

We report the system FPS in our paper as described above because iMAP, NICE-SLAM and Co-SLAM all share the same system design of running mapping every 5 frame. However, in Vox-Fusion they adopt a different strategy where mapping is performed as frequently as possible. That's why we reported separate FPS for tracking and mapping in our benchmark.

Now you should understand that the FPS we reported indicates the theoretical throughput a system can achieve. It doesn't reflect the actual run-time performance on a real machine. That's why we also reported the "total processing time" on ecah scenes under the breakdown section of each dataset.

random-guest, 2023-07-04:
@JingwenWang95 

Thank you very much for the detailed explanation.

Does it make sense to divide the total processing time for each scene by the number of frames to get the time a method needs to process a single frame (including tracking, mapping, and everything in between)?

closed 2023-07-06 by random-guest