HengyiWang/Co-SLAM #23: Some question about test result

zjulabwjt, 2023-07-22:
Thanks for your great work!I run your code in some dataset,and have some question.
1.About FPS mentioned in your main paper.I run your code in A40 Server,in replica dataset,about 2000 frames.The time consume about 20 min in your deault set(tracking iter 10,mapping iter 10).How to understand FPS mentioned in your paper?
2.I found in Real world dataset,the mesh result is bad,such as in Scannet and tum_rbd datasets.I found tracking and mapping iter is not enough,because the psnr is low.Can you give me some advice to improve the performance in real datasets？Add iters or higher voxel resolution but reduce efficient?
[image]

HengyiWang, 2023-07-22:
Hi @zjulabwjt, thank you for your questions. I'll address each of your concerns separately.

Regarding your first question, you can check  Jingwen's reply here https://github.com/HengyiWang/Co-SLAM/issues/11. Feel free to reach out if you cannot address this issue following those steps.

Regarding your second question:

>  Mesh is bad & PSNR is low -> tracking/mapping iterations are not enough -> asking suggestions for better reconstruction (giving two options that negatively impact efficiency: increase #iters or resolution).

1. Good or bad reconstruction on real-world datasets can depend on various aspects such as sensor quality, camera trajectories, illumination conditions, the effectiveness of the algorithms, subjective perceptions, etc. It is better to compare with other methods as those datasets are quite challenging. The screenshot you show is a reconstruction of the scene in the TUM dataset, which contains lots of unobserved regions and missing depth measurements. I have attached the results of KinectFusion for your reference. Our reconstruction usually has better completion (e.g., the screen.) BTW, if you want to get rid of floaters, simply run marching cubes by masking out points that are outside of the camera frustum.

[image]

2. Regarding PSNR, it's important to note that our primary focus is SLAM rather than novel view synthesis. Thus, instead of building a 3D radiance field, we mostly care about the color of the surface points no matter which view directions we use. So over-fitting on the color of each frame is not that necessary since the video contains many redundant observations of surface points (This is why we only keep 5% pixels of each keyframe to save CPU memory, and you will not expect a high re-rendered PSNR from only 5% pixels, but the actual color of the mesh is still comparable or sometimes even better than NICE-SLAM).  Whether #mapping iters is enough usually depends on the tracking performance in the context of SLAM as poor mapping would lead to lose tracking.

3. Adding those extra computational costs will give you slightly better results. However, our Co-SLAM aims to demonstrate improved performance while being computationally efficient compared to NICE-SLAM. Depending on your applications, I would suggest some other ways to improve the quality of the reconstruction.

zjulabwjt, 2023-07-23:
Hi @HengyiWang ,thank you for your quick and detailed reply and advice.

I have also some questions about the difference global BA in your code and local BA in other neural field SLAM.

As metioned in your main paper,the global BA select 5% ray in global keyframe set.But local BA only select less than 10 

keyframs in slid window.In my opinion,global BA can get more global consistency but less effience.Your Strategy in global BA 

(select 5% ray) aim to more global consistency result for tracking but decrease render perference? if local BA will increase render 

performance because select more rays in near view?

Thanks for your reply and help!

HengyiWang, 2023-07-23:
Hi @zjulabwjt, thank you for your questions.

Regarding the 5% pixels mentioned in the paper, it serves as a keyframe management strategy to optimize CPU memory usage by storing 5% pixels of each keyframe in the CPU. The key idea is that neural implicit SLAM doesn't require storing frame-level information for all pixels, and keeping 5% of keyframe pixels in CPU is often sufficient for the reconstruction. However, you can still increase the percentage up to 100% and performance will usually get better. This keyframe management strategy is independent of global BA.

As for local BA, it can definitely help with rendering performance as it samples more rays from unexplored areas. However, performing local BA usually requires you to freeze the decoder. Otherwise, you may have a catastrophic forgetting issue. Check out our project page for a comparison with concurrent works (ESLAM) and more discussions about global/local BA at https://github.com/HengyiWang/Co-SLAM/issues/19.

Feel free to ask if you have any further questions. Your support is highly appreciated, and if you find our work helpful for your research, we'd be thrilled if you could star this repository ;)

zjulabwjt, 2023-07-23:
> highly appreciated

Thanks for your quick reply!I am highly appreciated and I have stared your excellent work!

closed 2023-07-23 by zjulabwjt

zjulabwjt, 2023-07-23:
Hi,@HengyiWang I am sorry to bother you again.Actually,I also have some questions about code.

1、I found in scannet scene mesh performance is bad.I found in scannet scene_0059_00 the hash size is only 12,the small size of hash just for less memeory?

2、I remove the one blob encode and only hash encode in scannet scene_0000,scene_0059,scene_0106,scene_0169,and i found nearly no difference in mesh result.In my option,one blob might fill the code and smooth surface but might loss the geometry detail.

3、I have a silly question,how to  run marching cubes by masking out points that are outside of the camera frustum?If it is convenient for you,can you share me this part of the code.

Thank you in advance for your reply！

HengyiWang, 2023-07-23:
Hi @zjulabwjt, thanks for your question.

1. The description of "mesh is bad" is quite vague... Can you elaborate more about why you think it is bad or show some comparisons so that I can help you with it? The hash size is determined based on the size of the scene. You can tune the hash size and the smaller it is, the less memory you will use.

2. Coordinate encoding can give better hole-filling ability as shown in Fig.2 in our paper. However, in scannet, most of the time, the hole is quite small, and the continuity of the feature grid is usually more than enough. Additionally, joint encoding can help with memory compression. You can further reduce the hash look-up size and get something like Fig. 9 in our main paper. As for the concern about losing geometry details, we have not observed such issues, as sparse parametric encoding usually complements coordinate encoding and effectively preserves the geometry details

3. You can try mesh culling described here [neural_slam_eval](https://github.com/JingwenWang95/neural_slam_eval#evaluation-protocol). Using either frustum (NICE-SLAM and iMAP) or frustum+occlusion (Neural-RGBD and GO-Surf) strategy.

zjulabwjt, 2023-07-24:
Hi,@HengyiWang thank you again for your patient reply!
I use the pretrained model in Coslam with entire scene in scannet scene0059_00,and inference the whole image in sequence.But I found such result.Many pixels' color seems is error,it is because in SLAM pipline,only focus on a part of pixels that cause low rendering result(even global BA,sparse pixels in perivious frames)?I don't know it is reasonable(just before you said,your work is focus on tracking not rendering),I want to incremantal reconstruction scene with higher render performance,Can you give me some advice?
[image]

HengyiWang, 2023-07-24:
Hi @zjulabwjt, thanks for sharing the results, now I see what you mean. The rendering result might be affected by the following factors:

1. Insufficient pixel sampling: We have discussed this issue before.
2. Tracking accuracy: As far as I remember, the ATE is around 10cm on this scene, leading to thicker structures in the reconstruction

I have added one more function of current frame mapping in the latest commit. The decoder is frozen to ensure the local mapping will not ruin the optimisation of the decoder, which might be helpful for your concern and can also potentially address the concerns raised by @supremeccccc https://github.com/HengyiWang/Co-SLAM/issues/19.

You can try some of the following things:

1. Add some iterations for current frame mapping here: https://github.com/HengyiWang/Co-SLAM/blob/2329d09cc27133e710cf5e9ea1d9aaa4fa2b6662/configs/ScanNet/scannet.yaml#L13
2. Increase the percentage of pixels stored in the keyframe here (only increase the memory usage of CPU ): https://github.com/HengyiWang/Co-SLAM/blob/2329d09cc27133e710cf5e9ea1d9aaa4fa2b6662/configs/ScanNet/scannet.yaml#L20
3. Increase iters & samples for BA (It is always better to increase batch size first and then try to increase iters)

zjulabwjt, 2023-07-25:
> Hi @zjulabwjt, thanks for sharing the results, now I see what you mean. The rendering result might be affected by the following factors:
>
> 1. Insufficient pixel sampling: We have discussed this issue [...]

Thanks for your reply,And I will try your advice.Thanks for your excellent work and help again.

HengyiWang, 2023-07-25:
1. ScanNet pose is obtained by BundleFusion, which contains NaN. Check if the NaN is caused by pose. As far as I remember, scan_0059 has NaN pose.
2. Try increasing cur_frame_iters: 0 and see if it helps with PSNR in your case.

zjulabwjt, 2023-07-26:
> 1. ScanNet pose is obtained by BundleFusion, which contains NaN. Check if the NaN is caused by pose. As far as I remember, scan_0059 has NaN pose.
> 2. Try increasing cur_frame_iters: 0 and see if it helps with PSNR in your case.

Thanks for your help!I found in scene0059_00 dataset frames 1460-1472 is inf value and cause this problems.I will try to increase cur_frame_iters for better render performance.But I find the mesh result use tracking pose is better than use gt_pose in scannet scene0106_00 in visiualization using same config(cur_iter_frame is zero).
Why cause this result?If the pose_gt is not realiable?
the first image uses tracking pose and second one uses gt_pose,all meshs is culled.
[image]
[image]

HengyiWang, 2023-07-26:
Yes, the poses are obtained by BundleFusion, it is not strictly gt pose. For example, take a look at the gt mesh of scene0000, you can see lots of misaligned areas, which are caused by inaccurate pose obtained by BundleFusion.

zjulabwjt, 2023-07-26:
> Yes, the poses are obtained by BundleFusion, it is not strictly gt pose. For example, take a look at the gt mesh of scene0000, you can see lots of misaligned areas, which are caused by inaccurate pose obtained by BundleFusion.

Thanks for your reply!So the gt_pose is in inaccurate,the pose evaluation with est_pose if is not have reference value.Why neural slam paper use the not strictly gt pose for evaluation?

HengyiWang, 2023-07-26:
Because you will only get the 100% gt poses in a synthetic dataset.

zjulabwjt, 2023-10-12:
hi, @HengyiWang ,I have some quesutions about how to eval runtime and params?
1、about params compute,what specific python statements i should use?
if it is correct use blow statements?
`params = list(model.parameter())
num_params = sum([p.numel() for p in params])
print("num_params: {} M".format(num_params*4/(1024*1024)))`
2、how to accuratly compute every iter in mapping and tracking? what specific python statements i should use?
Thanks for your reply!

mohcenaouadj, 2024-06-25:
Hello @zjulabwjt, great work retrieving the rendered images, can you please share the script with me or at least explain to me the procedure to do so ?