HengyiWang/Co-SLAM #26: a question about ATE RMSE results on ScanNet

wangm-buaa, 2023-07-28:
Hi,  thanks for your excellent work!
I have a question after evaluating scene0000_00 with the code you provided. There are two types results `pose_x.png `and `pose_r_x.png` generated by following codes. 

https://github.com/HengyiWang/Co-SLAM/blob/2329d09cc27133e710cf5e9ea1d9aaa4fa2b6662/coslam.py#L675-L676

It seems that using `pose_relative` can achieve better ATE RMSE result,  while the other result is more similar as the one on paper.
I wonder which result is the one you provided on paper and why you didn't use the other one.
Thank you!

HengyiWang, 2023-07-28:
Hi @wangm-buaa, thank you for your interest in our work. **We use `pose_relative` in the paper**. The difference between those two is that in `pose_relative`, we save the relative poses of the non-keyframe with respect to the previous keyframe while in `est_c2w_data `, the non-keyframe poses are saved with respect to the first frame. Keyframe poses will be updated after performing BA, then all poses in `pose_relative` would change. Most of the time, the results using `pose_relative` are better meaning that BA is working very well.  Thus, I keep both results in the code :)

Please note that in ScanNet, we run each scene 5 times and report the average results. Also, I did some refactoring of the code in the released version, so you might get slightly better tracking results.

wangm-buaa, 2023-07-28:
Thanks for your reply and help!

closed 2023-08-16 by wangm-buaa

ArlenCHEN, 2023-09-25:
Following your discussions, is it for Co-SLAM to have better results if we use `pose_relative` instead of `est_c2w_data` to perform the ray tracing in `tracking_render` and `global_BA`?

HengyiWang, 2023-09-26:
Hi @ArlenCHEN, we only use the keyframe rays and the corresponding poses for global BA.

ArlenCHEN, 2023-09-27:
Thanks @HengyiWang.

Another question: How do you generate the online reconstruction video? Did you release the code for it?