HengyiWang/Co-SLAM #33: First frame initialization issues about pose estimation

Aurora-chevalier, 2023-09-11:
Hello Dr, I have questions about the position of your pose initialization, when I change the first frame of initialization prediction pose est_pose (0) to the given second frame gt pose, the reconstruction effect is very unsatisfactory, do you know what is the cause? Or does co-slam have to set the prediction est_pose (0) to the gt pose of the first frame to achieve pose estimation for subsequent frames?
[image]

HengyiWang, 2023-09-11:
Hi @Aurora-chevalier, this is because when you change the first frame pose, you need to have a corresponding scene bound that covers the entire scene. Co-SLAM does not require setting the first frame pose as the gt pose.

Aurora-chevalier, 2023-09-11:
Thank you very much for your reply, do you have any suggestions for solving the minimum bound after changing the pose of the first frame? Or is there any way to fix this?

ArlenCHEN, 2023-09-25:
Hi @HengyiWang, thanks for your explanation. 

I noticed that in the default code, the `first_frame_mapping` uses the gt pose. Will the results be worse if we do not use the gt pose, and use, say, randomly initialized pose values? I wonder if you have any investigations on this. Thanks!