HengyiWang/Co-SLAM #34: sdf_mask and fs_weight in get_masks

mian-zhi, 2023-10-07:
Hello Ph.D, in the get_masks function of Model / Utils.py, 
calculate
```
fs_weight` = 1.0 -num_fs_samples / num_samples
sdf_weight = 1.0 -num_sdf_samples / num_samples
```python
Is it inconsistent with the formula in the paper? Whether it should be:
```
fs_weight = num_fs_samples / num_samples
sdf_weight = num_sdf_samples / num_samples
```python

mian-zhi, 2023-10-07:
If all the sampling points fall in the cut -off area, according to the original code, then the SDF_Weight will be 0, and the fs_weight will be 1.

HengyiWang, 2023-10-07:
Hi @mian-zhi, this is a strategy for balancing the weight of SDF and free space. For each ray, most of the sampled points would be in free space, which will make the MLP biased toward free space. Thus, we need to use this weight to balance the SDF and free space loss.

mian-zhi, 2023-10-08:
I understand. Thank you, Doctor.

closed 2023-10-08 by mian-zhi

akawincent, 2023-10-08:
> Hello Ph.D, in the get_masks function of Model / Utils.py, calculate
> 
> ```
> fs_weight` = 1.0 -num_fs_samples / num_samples
> sdf_weight = 1.0 -num_sdf_samples / num_samples
> ```
> 
> Is it inconsistent with the formula in the paper? Whether it should be:
> 
> ```
> fs_weight = num_fs_samples / num_samples
> sdf_weight = num_sdf_samples / num_samples
> ```

That is a good question! 
It was also confused me. Thanks for your answer. @HengyiWang