HengyiWang/Co-SLAM #36: Floaters in the rendering visualizations

ming82871, 2023-11-07:
Hi!
Thanks for your great work! We intend to evaluate some Tum sequences with your code. We found that there are some floaters in the rendering image and depth(We provide some visualizations about fr2_xyz and fr3_long_office sequence). Can you give me some suggestions or experience about this phenomenon? Do you think Hash collision is a potential cause? 
Thanks!

[image]
[image]

HengyiWang, 2023-11-07:
Hi @ming82871, thanks for your interest in our work. There are several potential reasons for this issue. 1) For rendering these results, have you used depth-guided sampling? As our truncation region is quite small, you need to use either depth-guided sampling or importance sampling (or just a sufficiently large amount of sample points) to make sure there are sample points within the truncation region. 2) Co-SLAM keeps and trains on only a small fraction of pixels with a small number of iterations for efficiency, you may not be able to get a high PSNR for rendering due to the insufficient number of pixels used for training.

glaciomarine, 2023-11-20:
Hello, excuse me, how to do visualization

Hammond-Duck, 2025-05-19:
> Hi! Thanks for your great work! We intend to evaluate some Tum sequences with your code. We found that there are some floaters in the rendering image and depth(We provide some visualizations about [...]

May I ask you how to get the rendered images?