HengyiWang/Co-SLAM #37: (Environment installing concerned) error: command 'E:\\Microsoft Visual Studio\\2019\\Professional\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

saulgooodman, 2023-11-08:
Hello there. I was about to install the Co-SLAM environment, when I was installing the 'marching_cubes' according to the given README.md (which is cd external/NumpyMarchingCubes; python setup.py install) but I encountered a problem error: 
'command 'E:\\Microsoft Visual Studio\\2019\\Professional\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2'
I use windows platform and the python version is 3.8.
Could you please explain why? Tks!

HengyiWang, 2023-11-08:
Hi @saulgooodman, this error message does not provide any information about why this compilation error happens, can you try to check the compiler output for a more detailed error message so I can help you with it?

In the meantime, maybe check this issue: https://github.com/HengyiWang/Co-SLAM/issues/5 to see if this can help you to fix the problem.

saulgooodman, 2023-11-09:
issue #5 didn't solve my problem :<
The error is given below. Tks!
(pytorch_env) E:\桌面\Co-SLAM-main\external\NumpyMarchingCubes>python setup.py install
E:\anaconda\envs\pytorch_env\lib\site-packages\setuptools\installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(
running install
E:\anaconda\envs\pytorch_env\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
E:\anaconda\envs\pytorch_env\lib\site-packages\setuptools\command\easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
writing NumpyMarchingCubes.egg-info\PKG-INFO
writing dependency_links to NumpyMarchingCubes.egg-info\dependency_links.txt
writing top-level names to NumpyMarchingCubes.egg-info\top_level.txt
reading manifest file 'NumpyMarchingCubes.egg-info\SOURCES.txt'
writing manifest file 'NumpyMarchingCubes.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'marching_cubes._mcubes' extension
"E:\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Imarching_cubes/src -IC:\Users\fqy\AppData\Roaming\Python\Python38\site-packages\numpy\core\include -IE:\anaconda\envs\pytorch_env\include -IE:\anaconda\envs\pytorch_env\Include "-IE:\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IE:\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /EHsc /Tpmarching_cubes/src/_mcubes.cpp /Fobuild\temp.win-amd64-cpython-38\Release\marching_cubes/src/_mcubes.obj -std=c++11 -Wall
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
_mcubes.cpp
E:\anaconda\envs\pytorch_env\include\object.h(182): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::slot”后
E:\anaconda\envs\pytorch_env\include\object.h(190): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::flags”后
E:\anaconda\envs\pytorch_env\include\cpython/object.h(212): warning C4820: “_typeobject”:“4”字节填充添加在 数据成员“_typeobject::tp_flags”后
E:\anaconda\envs\pytorch_env\include\cpython/object.h(256): warning C4820: “_typeobject”:“4”字节填充添加在 数据成员“_typeobject::tp_version_tag”后
E:\anaconda\envs\pytorch_env\include\bytearrayobject.h(30): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::ob_exports”后
E:\anaconda\envs\pytorch_env\include\bytesobject.h(41): warning C4820: “<unnamed-tag>”:“7”字节填充添加在 数据成员“<unnamed-tag>::ob_sval”后
E:\anaconda\envs\pytorch_env\include\bytesobject.h(165): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::small_buffer”后
E:\anaconda\envs\pytorch_env\include\cpython/unicodeobject.h(221): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::state”后
E:\anaconda\envs\pytorch_env\include\cpython/unicodeobject.h(625): warning C4820: “<unnamed-tag>”:“2”字节填充添加在 数据成员“<unnamed-tag>::readonly”后
E:\anaconda\envs\pytorch_env\include\longintrepr.h(88): warning C4820: “_longobject”:“4”字节填充添加在 数据成员“_longobject::ob_digit”后
E:\anaconda\envs\pytorch_env\include\memoryobject.h(45): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::flags”后
E:\anaconda\envs\pytorch_env\include\memoryobject.h(62): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::flags”后
E:\anaconda\envs\pytorch_env\include\methodobject.h(56): warning C4820: “PyMethodDef”:“4”字节填充添加在 数据成员“PyMethodDef::ml_flags”后
E:\anaconda\envs\pytorch_env\include\moduleobject.h(63): warning C4820: “PyModuleDef_Slot”:“4”字节填充添加在 数据成员“PyModuleDef_Slot::slot”后
E:\anaconda\envs\pytorch_env\include\cpython/initconfig.h(16): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::_type”后
E:\anaconda\envs\pytorch_env\include\cpython/initconfig.h(19): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::exitcode”后
E:\anaconda\envs\pytorch_env\include\cpython/initconfig.h(202): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::parse_argv”后
E:\anaconda\envs\pytorch_env\include\cpython/initconfig.h(357): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::legacy_windows_stdio”后
E:\anaconda\envs\pytorch_env\include\cpython/initconfig.h(367): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::pathconfig_warnings”后
E:\anaconda\envs\pytorch_env\include\cpython/initconfig.h(374): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::module_search_paths_set”后
E:\anaconda\envs\pytorch_env\include\cpython/initconfig.h(393): warning C4820: “<unnamed-tag>”:“4”字节填充添加在 数据成员“<unnamed-tag>::skip_source_first_line”后
E:\anaconda\envs\pytorch_env\include\cpython/pystate.h(65): warning C4820: “_ts”:“2”字节填充添加在 数据成员“_ts::recursion_critical”后
E:\anaconda\envs\pytorch_env\include\cpython/pystate.h(73): warning C4820: “_ts”:“4”字节填充添加在 数据成员“_ts::use_tracing”后
[... cut ...]
marching_cubes/src/marching_cubes.cpp(459): warning C4365: “参数”: 从“int”转换到“const unsigned __int64”，有符号/无符号不匹配
marching_cubes/src/marching_cubes.cpp(459): warning C4365: “参数”: 从“int”转换到“const unsigned __int64”，有符号/无符号不匹配
marching_cubes/src/marching_cubes.cpp(459): warning C4365: “=”: 从“int”转换到“_Ty”，有符号/无符号不匹配
        with
        [
            _Ty=unsigned long
        ]
error: command 'E:\\Microsoft Visual Studio\\2019\\Professional\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

HengyiWang, 2023-11-09:
Hi @saulgooodman, it looks like this error is likely to be related to your system and compiler. As I am not familar with compilation using Windows, I would suggest double-checking something like the compiler version or Visual Studio installation.

Pallab007Saha, 2024-02-22:
Hii @saulgooodman @HengyiWang , were you able to find any solution for the above error ?

ShunkaiZhou, 2024-05-15:
Hi, I am experiencing the same problem, has it been resolved please?

> 嗨，您能找到解决上述错误的方法吗？在其他论坛中也存在类似的错误，但在这种情况下没有可用的孤独解决它。