HengyiWang/Co-SLAM #39: Help Needed with Custom RealSense Dataset and Debugging

Orwlit, 2023-11-21:
First off, huge thanks for all the hard work on this repo! It's been super helpful, and I'm really grateful for what you've put together here.

I'm diving into recording my own RealSense dataset and I'm hoping to get it working with your code. Got a couple of questions I'm scratching my head over, and I'd love your input:

**1. About `my_room` dataset:**
- What's the deal with `align_mat.txt` and `trajectory.log` in the dataset? What do they actually do?
- Would it be possible for you to share the YAML config file you used for your "my_room" RealSense D435i dataset? It'd be super helpful to see how you set it up. Thanks a bunch!
- If I want to make these files for my own dataset, any tips on how to go about it? Are there any tools or methods you'd recommend?

**2. Debugging my Dataset:**
- I'm trying to debug some frames to see how the network is rendering the RGB and depth images. Any pointers on which part of the code I should dive into for this kind of info?

Any advice or pointers you can share would be awesome. Thanks a ton in advance!

HamzaOuajhain, 2024-04-06:
I would like to create my own dataset using RealSense too, have you found any cues about the issues you encountered ?

Lucasmogsan, 2024-09-05:
Any updates on this @HengyiWang ?