HengyiWang/Co-SLAM #40: How to set parameters for NICE-SLAM identical comparison

anghaocheng, 2023-11-23:
Thanks so much for sharing this excellent work. It has excellent reconstruction effect, but I set it up as in the supplementary material of the article
hash_size: 14
smooth_weight: 0.001
Then run the code on office0 and sample the generated results using the same comparison method as nice-slam, and get
accuracy: 9.465963119058479
completion: 1.6472761277591914
completion ratio: 96.2565
Why is accuracy so ridiculously high?

HengyiWang, 2023-11-23:
Hi @anghaocheng, I have tested the code on office0 just now. The results look good. Can you check if you set up the hyper-parameters correctly?

High accuracy is usually because if you look into the mesh reconstruction, you can see floaters outside the inner surface of the scene, caused by the continuity of the scene representation. Those floaters that are outside of the ROI would still be sampled for evaluation and thus cause high accuracy (The comp and comp. ratio would not be affected).

anghaocheng, 2023-11-24:
Hi, thanks for your quick reply. 
I repeated the test and tested it in other scenarios in Replica. Although the accuracy is sometimes still slightly large (such as 2.2, 2.8, 3.4), there is no situation where the accuracy is  ridiculously  high. Perhaps the result about office0 above happened to be that the random error was too large at one time.

closed 2023-11-24 by anghaocheng