HengyiWang/Co-SLAM #41: Pose evaluation graph issues using Replica Dataset Office_0

saulgooodman, 2023-12-06:
When I was using Replica Dataset Office_0 as the dataset, I found the predicted trajectory map and the ground truth trajectory map exhibit similar shapes but differ significantly in scale, with an approximate discrepancy of 5x.
Could you please explain why this happened? Is this issue happening during drawing using Matplotlib, or is it due to errors in pose estimation during Bundle Adjustment (BA)? In other words, does this plotting with Matplotlib indicate a potential impact on SLAM performance?
Thanks a lot!
[image]

HengyiWang, 2023-12-06:
Hi @saulgooodman, I am not sure what is going on here. I have never observed this issue before. Can you check if you downloaded everything correctly and have not changed any configs?

saulgooodman, 2023-12-08:
Yeah, I have downloaded everything correctly and I used the default config (except num_workers changed from 4 to 0 but it shouldn't have any effect on plotting the graph ). BTW I tested PSNR of the rendering result (PSNR=27.29) , the PSNR seems OK.

HengyiWang, 2023-12-10:
Can you check whether the reconstructed mesh aligns with the ground-truth mesh (Without any significant difference in scale)? Additionally, please also make sure you turn on the SDF loss or depth loss.

saulgooodman, 2023-12-18:
SDF loss and depth loss are turned on. However there exists a scale error between GT and the output :(   
[image]

HengyiWang, 2023-12-18:
That is so weird. I have never experienced this behavior before. Can you check your data (depth map) and see if you have set up a correct depth scale in the config file? I guess there must be something wrong with the data, or configs in your case. Maybe you can follow the instructions in the README to download the repository as well as the replica scene again and re-run Co-SLAM if you still cannot figure out what caused this issue.

saulgooodman, 2023-12-20:
I chose another version of Replica and the scale returned to normal! Thanks to your patience! :)

closed 2023-12-20 by saulgooodman