HengyiWang/Co-SLAM #47: PolyCam dataset

mohcenaouadj, 2024-05-23:
Hello, 

Great work, I really appreciate everything about this project.

I was wondering concerning the iphone dataset, I have an RGB-D dataset captured by [PolyCam ](https://poly.cam/) on Iphone with image data in shape of 1024*768 and depth data in shape 256*192.

I'm trying to use the iphone configuration with this dataset yet for some reason I receive some really bad representation and trajectory, I was wondering what are the parts of the configuration that you recommend working on, mainly I only changed the camera size and the intrinsics.

I tried also to use vis_bound script to generate the corresponding bounding box, but the output is always: TriangleMesh with 0 points and 0 triangles.

[image]

HengyiWang, 2024-05-23:
Hi @mohcenaouadj, thank you for your question. For running Co-SLAM on iPhone datasets, here are some tips:

1. Make sure you are using the correct intrinsic & depth scale. You can check it using https://github.com/HengyiWang/Co-SLAM/blob/main/vis_bound.ipynb. If everything is good, you are expected to see the fused scene. Based on your description, I guess you may not set the depth scale correctly (It should be 1000 or something. You need to double-check with Polycam). 
2. If you use identity to initialize the pose, make sure to use quaternion as the rotation representation
3. It is suggested to upsample the depth to 512x384 to give denser supervision to the scene representation

mohcenaouadj, 2024-05-24:
@HengyiWang Thank you so much for your reply, I really appreciate it.

I'm still having the same problem, I fixed the intrinsics and the depth scale, which are provided by the app itself, and double-checked using Colmap, yet even the script **vis_bound** isn't giving good results, although when I tried it on *Replica* dataset, the result was perfect, so I was wondering what maybe other parameters that can be considered responsible for this.

Vis Bound result :
[image]

Coslam result : 
[image]

Thanks again for your help !

HengyiWang, 2024-05-24:
Hi @mohcenaouadj, can you check the camera pose (c2w or w2c, if it is w2c, you need to inverse it in your dataset class) and its convention (OpenCV or OpenGL, we use GL here)

mohcenaouadj, 2024-05-28:
Hello again @HengyiWang, I have now used StrayScanner to help me understand the project more, the results were awesome, and I really thank you for this project.

What I understood from this experience is that even PolyCam intrinsics should be divided by factor, like you did with StrayScanner "**7.5**" ... I tried 7.5 for PolyCam and didn't give results, so I was wondering how did you know that that was the scale to divide on? is there a math formula or something?

HengyiWang, 2024-05-28:
Glad to hear! 7.5 should be the scale factor between RGB and depth map. For Polycam, you may want to check this repo (https://github.com/PolyCam/polyform/) to check their conventions. From: Mohcen AOUADJ ***@***.***>Date: Tuesday, May 28, 2024 at 8:08 AMTo: HengyiWang/Co-SLAM ***@***.***>Cc: wanghy ***@***.***>, Mention ***@***.***>Subject: Re: [HengyiWang/Co-SLAM] PolyCam dataset (Issue #47)Hello again @HengyiWang, I have now used StrayScanner to help me understand the project more, the results were awesome, and I really thank you for this project.What I understood from this experience is that even PolyCam intrinsics should be divided by factor, like you did with StrayScanner "7.5" ... I tried 7.5 for PolyCam and didn't give results, so I was wondering how did you know that that was the scale to divide on? is there a math formula or something?

HengyiWang, 2024-05-30:
1.  The GPU memory usage mostly depends on how many sample points you have
2. We have several voxel sizes in the config file. `voxel_sdf` and `voxel_rgb` are voxel sizes of sdf and color hash grid. By setting `oneGrid: True`, you only have one SDF feature grid. We also have `voxel_eval` and `voxel_final: 0.03`, which are the size used to extract the mesh. You can use a smaller voxel_final to extract mesh with higher resolution. In the meantime, you may want to tune the size of `voxel_sdf` to achieve better performance.

mohcenaouadj, 2024-06-04:
Hello again @HengyiWang, 

When it comes to mesh resolution, especially small objects, I still suffer from low quality, and I believe that comes from the fact that we downsampling the RGB image, so we lose a lot of details, I was wondering if you're familiar with practical depth upsampling technique that might help me solve this problem ?

Thanks again !

closed 2024-06-17 by mohcenaouadj