HengyiWang/Co-SLAM #50: Tracking Error

mohcenaouadj, 2024-07-02:
Hello Again,

The results you see is generated using Co-SLAM, of course i used object segmentation and confidence maps and played a little bit with sampling methods and other hyperparameters, yet the one on the right is generated using only the first frame, and the second is the whole mesh using 200 frames, our explanation is that the problem is in the pose estimation, that's why we have such bad quality at the end.

What do you propose as modification or adjustment in the hyperparameters in order to get better tracking ?
[image]
[image]

Thanks again !

HengyiWang, 2024-07-03:
I would suggest doing a visualization of online mapping or just saving mesh every 50 frames to see what is going on. As you need a detailed reconstruction with 200 frames, you might want to increase the #pixels in the keyframe database instead of just using 5%.  (btw, are these two meshes extracted using the same voxel size?)

mohcenaouadj, 2024-07-08:
Hello, here some notes I took that I wish to share with you: 
- I have used the mesh saving technique, I tried to calculate the PSNR as well as retrieve the rendered the images to observe the problem, the PSNR starts at 37 and just by the next frame it decreases to 19 and never surpasses 21 for the rest of the training.
 
[image]
[image]
[image]
[image]
[image]

You can see that in the first frame there's a lot of noise related to the SDF which gets cleaned by time, yet the resolution also is lost.
- I have to note also that I used about 90% of pixels in the database, and 100 iters for mapping and 50 for tracking.
- My sampling technique is based on confidence maps and segmentation masks, I first apply the segmentation mask on the rgb, depth and confidence map, then what rest from the information, I use the confidence map to guide the sampling, 70% from the high confidence regions, and 30 for the mid regions. yet it's true, when it comes the sampling that is made to store the pixels in the database, I haven't address that yet, which I believe it can help the resolution a little bit more, but still, I don't believe it's enough.
- Do you have any suggestion on how to use Co-SLAM without the depth maps, or at least lower the dependence on them?
- What is exactly the use of this parameter: **config['training']['trunc']** ? is it to filter SDF collisions? I'm using 0.05 which I believe it's 5cm, less than that we start seeing bold shapes and edges, less we start losing the object, does it have to do with the quality of the rendered depth maps? in other words 

I hope you can help me with this.
Thank you

HengyiWang, 2024-07-11:
Yes, that makes sense because your first frame pose would not be changed so you basically just overfitting on it. In that case, the PSNR will be high. And PSNR is a metric that is super sensitive to the inaccuracy of the camera pose (Maybe also look into SSIM, LPIPs). For tracking, I think in MonoGS (https://arxiv.org/pdf/2312.06741, https://github.com/muskie82/MonoGS) they have done an analysis on Neural SDF, you may want to have a look.

HengyiWang, 2024-07-11:
If you want to have a lower dependence on depth, you may need to change the rendering pipeline (VolSDF for instance). For questions regarding the truncation distance, check this thread and the related links https://github.com/HengyiWang/Co-SLAM/issues/2

closed 2024-07-22 by mohcenaouadj