HengyiWang/Co-SLAM #51: How to calculate total model parameter of Co-SLAM?

saulgooodman, 2024-07-03:
Hello, I tried to calculate the total parameter of Co-SLAM by calculating all the parameters in decoder, but I found the result is much smaller than the data in the paper. I thought it was because I didn't take the learnable parameters outside the decoder into consideration. I'd like to ask how you calculate all the parameter in Co-SLAM. Thanks a lot!

HengyiWang, 2024-07-11:
As far as I remember, we did not count the parameters for camera pose in the paper. But, yes, definitely decoder + hash grid. For example, let us say if your hash size is 10, then the total params for your hash grid should be something around 16 * 2^10 *2. where 16 is #resolution,  and 2 is feature dim.

FDF521X, 2024-10-21:
Why did you use hash=16 for experiments, but hash=13 for calculating the parameter count?

HengyiWang, 2024-10-21:
Sorry, I do not get what you mean. For Tab.1, all results are generated using hash=13. And 16 in my previous reply corresponds to the num of resolutions since we use a multi-resolution hash grid as in instant-ngp.

FDF521X, 2024-10-24:
> Sorry, I do not get what you mean. For Tab.1, all results are generated using hash=13. And 16 in my previous reply corresponds to the num of resolutions since we use a multi-resolution hash grid as in instant-ngp.

I apologize for not being clear. I noticed that in your provided code, the hashsize for the replica scenario was set to 16, so I thought this was your default configuration.