HengyiWang/Co-SLAM #53: Depth sampling and volume reconstruction

mohcenaouadj, 2024-07-23:
Hello,

In scene representation, depth is sampled using "mainly" two concatenated grids, one around the depth value itself and one alongside all the bounding box axis, I understand that the second one is useful in large scene (interior scenes) where it's important to note the free space between one object to another, yet one it comes to single object reconstruction, where we're trying to reconstruct one object only, supposedly we have masks around it and apply the pixel sampling on the object only ... etc, when we remove the samples alongside the bounding box axis, the whole volume gets filled with density points that mcubes will eventually take in consideration and object will disappears when I reconstruct the mesh !

Do you suggest any solution for issue?

HengyiWang, 2024-07-23:
Hi @mohcenaouadj,

That is a great question. In fact, since you already know that you only have one object around the center, you can do a good initialization of the SDF field. In our new paper MorpheuS, we initialize the SDF into an approximate sphere using something similar to https://arxiv.org/abs/2002.10099. I think you can try this implicit initialization or just use a fixed blob function and let the model fit the SDF residual.

mohcenaouadj, 2024-07-23:
@HengyiWang thank you a lot for your answer, I would like - if you have time - to give me a brief explanation of this phenomena, like to understand more the global behavior of Co-SLAM. 
It looks like the joint-coordinate model fills the volume since the beginning with density points, and then works its way to find objects and free space, if you can explain to me this part, I would be grateful.

HengyiWang, 2024-07-25:
This is because the coordinate encoding is continuous. With the inherent smoothness of MLPs, once you start to train the model, it can have some sort of extrapolation (It naturally fits indoor scene reconstruction as most parts of the scene are flat (floor, wall, etc).

mohcenaouadj, 2024-07-26:
Hello again @HengyiWang, Thanks a lot. the idea of initialization helped a lot, although I didn't use it as you proposed, yet it was hella useful, so thank you very much.

Concerning the free space, I noticed that you calculate the free space loss using only the front mask: 
https://github.com/HengyiWang/Co-SLAM/blob/dfb69c80beff76492309ffa626c5865ba8d8f0be/model/utils.py#L142

I was wondering how we can use the back mask as well in this situation? since we know we only focus on one central object; I believe it can be a good idea to use the back mask.

closed 2024-08-05 by mohcenaouadj