HengyiWang/Co-SLAM #59: Pose problem of Scannet dataset

atat1010, 2025-07-03:
Hi!
Thanks for your excellent work!I have a question about the Scannet dataset. After I obtained the dataset and extracted it, I found that the poses of some frames of the scenes used in the paper were inf. How did you deal with this?

HengyiWang, 2025-07-03:
Hi @atat1010, they will be masked during evaluation. Please check: https://github.com/HengyiWang/Co-SLAM/blob/3bb904e1c01bb3fd238a279e5ed3cdd291bb23b7/tools/eval_ate.py#L264

atat1010, 2025-07-04:
@HengyiWang 
Thank you very much for your reply, but I found that the sizes of ScanNet's RGB image and depth image are inconsistent, which seems to be caused by inconsistent camera intrinsic parameters. I checked the official github issues and it said that a simple reshape can be used, but the result I ran was very poor. Is there any other solution?